qqqlab/madflight #2: Unable to recognize MPU6050

01Space, 2023-12-14:
Hello!
this project is fantastic and I am very interested in it! I tried using RP2040 and ESP32 to test your code, but couldn't recognize MPU6050
I tested MPU6050 using other code and the hardware is working properly
I only connected to MPU6050, I did not connect to RC receiver
//========================================================================================================================//
//                                                 INCLUDES                                                               //
//========================================================================================================================//
//include hardware specific code
#if defined ARDUINO_ARCH_ESP32
  #include "hw_ESP32.h"
#elif defined ARDUINO_ARCH_RP2040
  #include "hw_RP2040.h"
  #include <FreeRTOS.h>  //FreeRTOS
  #include <semphr.h>    //FreeRTOS
#else 
  #error "Unknown hardware architecture"
#endif

#include "ahrs.h"

//========================================================================================================================//
//                                                 USER-SPECIFIED DEFINES                                                 //
//========================================================================================================================//

//-------------------------------------
// APPLICATION SETTINGS
//-------------------------------------
#define USE_IMU_INTERRUPT //Uncomment to use interrupt pin and not loop() to trigger loop_imu()

//-------------------------------------
// RC RECEIVER
//-------------------------------------
//Uncomment only one receiver type
#define USE_RCIN_PPM
//#define USE_RCIN_PWM 
//#define USE_RCIN_SBUS
//#define USE_RCIN_DSM
#include "src/RCIN/RCIN.h" //first define USE_RCIN_xxx then include RCIN.h

//-------------------------------------
// IMU SENSOR
//-------------------------------------
//Uncomment only one IMU
#define USE_IMU_MPU6050_I2C  //acc/gyro
//#define USE_IMU_MPU9150_I2C  //acc/gyro/mag
//#define USE_IMU_MPU6500_I2C  //acc/gyro      Note: SPI interface is faster
//#define USE_IMU_MPU9250_I2C  //acc/gyro/mag  Note: SPI interface is faster
//#define USE_IMU_MPU6500_SPI  //acc/gyro
//#define USE_IMU_MPU9250_SPI  //acc/gyro/mag

//Uncomment one I2C address. If unknown, see output of print_i2c_scan()
#define IMU_I2C_ADR 0x68 //MPU9250
//#define IMU_I2C_ADR 0x77 //MPU9150

//Full scale gyro range in deg/sec. Most IMUs support 250,500,1000,2000. Can use any value here, driver will pick next greater setting.
#define IMU_GYRO_DPS 250

//Full scale gyro accelerometer in G's. Most IMUs support 2,4,8,16. Can use any value here, driver will pick next greater setting.
#define IMU_ACCEL_G 2

//Uncomment only one sensor orientation. The labels is yaw / roll (in that order) needed to rotate the sensor from it's normal position to it's mounted position.
//if not sure what is needed: try each setting until roll-right gives positive ahrs_roll, pitch-up gives positive ahrs_pitch, and yaw-right gives increasing ahrs_yaw
//#define IMU_ROTATE_NONE
#define IMU_ROTATE_YAW90
//#define IMU_ROTATE_YAW180
//#define IMU_ROTATE_YAW270
//#define IMU_ROTATE_ROLL180
//#define IMU_ROTATE_YAW90_ROLL180
//#define IMU_ROTATE_YAW180_ROLL180
//#define IMU_ROTATE_YAW270_ROLL180
[... 128 lines cut ...]
E (4348) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=48
E (4348) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=97
E (4360) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=195
E (4371) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=48
E (4383) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=97
E (4394) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=195
E (4405) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=48
E (4417) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=97
E (4428) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=195
E (4440) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=48
E (4451) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=97
E (4462) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=195
float AccErrorX = +0.000000;
float AccErrorY = +0.000000;
float AccErrorZ = -1.000000;
float GyroErrorX = +0.000000;
float GyroErrorY = +0.000000;
float GyroErrorZ = +0.000000;
Paste these values in user specified variables section and comment out calculate_IMU_error() in void setup.
ahrs_Setup() No Magnetometer
ahrs_Setup() Estimated yaw:-0.00
gx:+0.00	gy:+0.00	gz:+0.00	ax:+0.00	ay:+0.00	az:+0.00	mx:+0.00	my:+0.00	mz:+0.00	roll:+0.0	pitch:-0.0	yaw:+0.0	yaw_mag:-0.0	imu_err_cnt:0	
gx:+0.00	gy:+0.00	gz:+0.00	ax:+0.00	ay:+0.00	az:+0.00	mx:+0.00	my:+0.00	mz:+0.00	roll:+0.0	pitch:-0.0	yaw:+0.0	yaw_mag:-0.0	imu_err_cnt:0	

Thank you!

qqqlab, 2023-12-14:
To which pins did you connnect the MPU6050? It should be SDA=23 SCL=22, or adapt hw_RP2040.h to match your pins.

qqqlab, 2023-12-14:
Apparently the output is generated with an ESP32. In the I2C scan it says `I2C: Found 112 device(s)` which is an indication that SDA and SCL are not connected at all. Connect SDA to pin 23 and  SCL to pin 22 and the I2C scan should report a single device. Use the reported I2C address as `#define IMU_I2C_ADR` in madflight.ino. Now the IMU sensor should work. 

I'm closing this issue. Please open a new issue, including your connections if you still have problems.

closed 2023-12-14 by qqqlab

01Space, 2023-12-14:
yes,ESP32  SDA=23 SCL=22 imu_INT_PIN = 39
RP2040
 i2c_SDA_PIN  = 20; 
 i2c_SCL_PIN  = 21
INT_PIN  = 22
I carefully checked the hardware cable connections and they are all correct