qqqlab/madflight #3: There may be an issue with the I2C code

01Space, 2023-12-14:
I modified the print_ I2c_ Scan() section code ，I modified the I2C PINS and now I can find the I2C device. So now I can confirm that the software is not working properly
void print_i2c_scan() {
  //Serial.printf("I2C: Scanning ...\n");
 // byte count = 0;
  //i2c->begin();
  //for (byte i = 8; i < 120; i++) {
   // i2c->beginTransmission(i);          // Begin I2C transmission Address (i)
   // if (i2c->endTransmission() == 0) { // Receive 0 = success (ACK response) 
    //  Serial.printf("I2C: Found address: 0x%02X (%d)\n",i,i);
   //   count++;
   // }
  //}
  //Serial.printf("I2C: Found %d device(s)\n", count);     
byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    WIRE.beginTransmission(address);
    error = WIRE.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

   
}
madflight starting 5 ...
madflight starting 4 ...
madflight starting 3 ...
madflight starting 2 ...
madflight starting 1 ...
USE_HW_ESP32
I2C: SDA=21 SCL=22
SPI: MOSI=23 MISO=19 SCLK=18 CS=15
Scanning...
I2C device found at address 0x68  !
done

USE_RX_PPM pin=35
USE_IMU_MPU6050_I2C
E (4065) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=48
E (4066) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=97
E (4077) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=195
E (4089) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=48
E (4100) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=97
E (4111) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=195
E (4123) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=48
E (4134) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=97
E (4145) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=195
E (4157) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=48
E (4168) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=97
E (4180) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=195
float AccErrorX = +0.000061;
float AccErrorY = +0.000061;
float AccErrorZ = -1.000061;
float GyroErrorX = -0.007634;
float GyroErrorY = -0.007634;
float GyroErrorZ = +0.007634;
Paste these values in user specified variables section and comment out calculate_IMU_error() in void setup.
ahrs_Setup() No Magnetometer
ahrs_Setup() Estimated yaw:-0.00
gx:+0.00	gy:+0.00	gz:-0.00	ax:+0.00	ay:+0.00	az:-0.00	mx:+0.00	my:+0.00	mz:+0.00	roll:+0.0	pitch:-0.0	yaw:-0.0	yaw_mag:-0.0	imu_err_cnt:0	
gx:+0.00	gy:+0.00	gz:-0.00	ax:+0.00	ay:+0.00	az:-0.00	mx:+0.00

01Space, 2023-12-14:
I used the code here
https://learn.adafruit.com/scanning-i2c-addresses/arduino

qqqlab, 2023-12-14:
Hi, thanks for trying this out. 

As I understand it, you changed the SDA pin from 23 to 21 in hw_ESP32.h.

But did you also change the HW_PIN_SPI_MOSI pin from 21 to 23??? Otherwise you have pin 21 defined twice, that does not work.

You should have something like this
```
//I2C:
const int HW_PIN_I2C_SDA  = 21; //<<<<<<<<<<<<<<<<<< was 23
const int HW_PIN_I2C_SCL  = 22; //default: Wire 22
typedef SoftWire HW_WIRETYPE; //define the class to use for I2C
//TwoWire *i2c = &Wire; //&Wire or &Wire1 - when using <Wire.h>
HW_WIRETYPE *i2c = new HW_WIRETYPE();  //create a ESP32_SoftWire instance

//SPI:
const int HW_PIN_SPI_MOSI = 23; //<<<<<<<<<<<<<<<<<<<<<< was 21
const int HW_PIN_SPI_MISO = 36; //VP defaults: VSPI 19, HSPI 12
const int HW_PIN_SPI_SCLK = 19; //   defaults: VSPI 18, HSPI 14
const int HW_PIN_SPI_CS   = 18; //   defaults: VSPI  5, HSPI 15
SPIClass *spi = new SPIClass(HSPI); // VSPI or HSPI(default)
```

Maybe try a fresh git clone, I've been making quite a few updates the last couple of days. Here is the output I'm getting on an ESP32 with MPU9150 (= MPU6050 + magnetometer) connected.

```
USE_HW_ESP32
I2C: SDA=21 SCL=22
SPI: MOSI=23 MISO=36 SCLK=19 CS=18
I2C: Scanning ...
I2C: Found address: 0x69 (105)
I2C: Found address: 0x77 (119)
I2C: Found 2 device(s)
USE_RCIN_CRSF
USE_IMU_MPU6050_I2C
....
CH1:1500	rcin_roll:+0.00	gx:-7.00	ax:-0.00	mx:+0.00	ahrs_roll:-0.1	roll_PID:+0.004	m1%:0	loop_rt:375	
CH1:1500	rcin_roll:+0.00	gx:-7.00	ax:-0.00	mx:+0.00	ahrs_roll:-0.2	roll_PID:+0.004	m1%:0	loop_rt:354	
CH1:1500	rcin_roll:+0.00	gx:-7.21	ax:-0.01	mx:+0.00	ahrs_roll:-0.2	roll_PID:+0.004	m1%:0	loop_rt:342	
```

01Space, 2023-12-15:
Thank you! Pin 21 is only defined once ，MPU9150 (=MPU6500+magnetometer), so MPU9250 and MPU6050 are completely different，I will test MPU6500 immediately
//I2C:
const int i2c_SDA_PIN  = 21; //default: Wire 21
const int i2c_SCL_PIN  = 22; //default: Wire 22
typedef SoftWire HW_WIRETYPE; //define the class to use for I2C
//TwoWire *i2c = &Wire; //&Wire or &Wire1 - when using <Wire.h>
HW_WIRETYPE *i2c = new HW_WIRETYPE();  //create a ESP32_SoftWire instance

//SPI:
const int spi_MOSI_PIN = 23; //   default: VSPI 23, HSPI 13
const int spi_MISO_PIN = 19; //VP default: VSPI 19, HSPI 12
const int spi_SCLK_PIN = 18; //   default: VSPI 18, HSPI 14
const int spi_CS_PIN   = 15; //   default: VSPI  5, HSPI 15
SPIClass *spi = new SPIClass(VSPI); // VSPI(default) or HSPI

01Space, 2023-12-15:
I tested using MPU6500, but your code still cannot recognize the device
I am using the ARDUINO1.8.16IDE and I have used the development board linked below
NodeMCU ESP32 S Dev Kit 
https://www.az-delivery.de/products/nodemcu-esp-32s-kit?shpxid=f07075dc-6a71-4466-9b0e-27f241a6759f
But I use RP2040+MPU6500, and your code can identify the device
madflight starting 5 ...
madflight starting 4 ...
madflight starting 3 ...
madflight starting 2 ...
madflight starting 1 ...
USE_HW_RP2040
I2C: SDA=20 SCL=21
SPI: MOSI=19 MISO=16 SCLK=18 CS=17
I2C: Scanning ...
I2C: Found address: 0x68 (104)
I2C: Found 1 device(s)
USE_RX_PPM pin=1
USE_IMU_MPU9250_I2C

01Space, 2023-12-15:
I used the code you updated today, RP2040 can recognize MPU6050 and MPU6500, but ESP32 cannot recognize MPU6050 and MPU6500.
RP2040 can only recognize that the device has no XYZ data output when connected to MPU6050. RP2040 can recognize that the device has XYZ data output when connected to MPU6500.

RP2040 connected to MPU6050

madflight starting 3 ...
madflight starting 2 ...
madflight starting 1 ...
USE_HW_RP2040
HW_PIN_LED=25
HW_PIN_RCIN_RX=1 TX=0
HW_PIN_IMU_INT=22
HW_PIN_I2C_SDA=20 SCL=21
HW_PIN_SPI_MOSI=19 MISO=16 SCLK=18 CS=17
HW_PIN_OUT[14]=2,3,4,5,6,7,8,9,10,11,12,13,14,15
I2C: Scanning ...
I2C: Found address: 0x68 (104)
I2C: Found 1 device(s)
USE_RCIN_CRSF
USE_IMU_MPU6050_I2C

RP2040 connected to MPU6500.

madflight starting 5 ...
madflight starting 4 ...
madflight starting 3 ...
madflight starting 2 ...
madflight starting 1 ...
USE_HW_RP2040
HW_PIN_LED=25
HW_PIN_RCIN_RX=1 TX=0
HW_PIN_IMU_INT=22
HW_PIN_I2C_SDA=20 SCL=21
HW_PIN_SPI_MOSI=19 MISO=16 SCLK=18 CS=17
HW_PIN_OUT[14]=2,3,4,5,6,7,8,9,10,11,12,13,14,15
I2C: Scanning ...
I2C: Found address: 0x68 (104)
I2C: Found 1 device(s)
USE_RCIN_CRSF
USE_IMU_MPU9250_I2C
IMU init failed rv=-1112. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
USE_IMU_MPU9250_I2C
IMU init failed rv=-1000. Retrying...
ahrs_Setup() Estimated yaw:-180.00
CH1:1459	rcin_roll:+0.00	gx:-95.15	ax:-0.38	mx:+0.00	ahrs_roll:-0.1	roll_PID:+0.048	m1%:0	loop_rt:891	gx:-95.15	gy:-107.25	gz:+125.13	ax:-0.38	ay:+0.13	az:-0.18	mx:+0.00	my:+0.00	mz:+0.00	roll:-0.1	pitch:-0.1	yaw:-179.9	yaw_mag:-0.0	
CH1:1500	rcin_roll:+0.00	gx:+25.71	ax:-1.29	mx:+0.00	ahrs_roll:-0.5	roll_PID:-0.012	m1%:0	loop_rt:564	gx:+25.71	gy:-8.97	gz:-8.10	ax:-1.29	ay:-1.32	az:+0.96	mx:+0.00	my:+0.30	mz:+0.00	roll:-0.5	pitch:-0.3	yaw:-179.3	yaw_mag:-90.0	
CH1:1500	rcin_roll:+0.00	gx:+25.91	ax:-1.29	mx:+0.00	ahrs_roll:-0.4	roll_PID:-0.012	m1%:0	loop_rt:646	gx:+25.91	gy:-8.81	gz:-8.32	ax:-1.29	ay:-1.33	az:+0.97	mx:+0.00	my:+0.30	mz:+0.00	roll:-0.4	pitch:+0.0	yaw:-179.4	yaw_mag:-90.0	
CH1:1500	rcin_roll:+0.00	gx:+25.91	ax:-1.29	mx:+0.00	ahrs_roll:-0.2	roll_PID:-0.012	m1%:0	loop_rt:663	gx:+25.91	gy:-9.04	gz:-8.11	ax:-1.29	ay:-1.32	az:+0.97	mx:+0.00	my:+0.30	mz:+0.00	roll:-0.2	pitch:-0.2	yaw:-178.8	yaw_mag:-90.0	
CH1:1500	rcin_roll:+0.00	gx:+25.91	ax:-1.29	mx:+0.00	ahrs_roll:-0.1	roll_PID:-0.013	m1%:0	loop_rt:673	gx:+25.91	gy:-8.81	gz:-8.32	ax:-1.29	ay:-1.33	az:+0.97	mx:+0.00	my:+0.30	mz:+0.00	roll:-0.1	pitch:+0.1	yaw:-178.

qqqlab, 2023-12-15:
I've updated imu.h, RP2040 connected to MPU6500 should work now.

"NodeMCU ESP32 S Dev Kit" - I'm not 100% sure this is a ESP32 or a ESP32-Sx variant, or some sort of fake. I've never seen the ESP32S module on the picture. 

Anyway, I've added `#define USE_ESP32_SOFTWIRE` in hw_ESP32.h, comment this line to use the regular Wire.h library. Maybe it helps.

Please give it a try, and let me know.

01Space, 2023-12-15:
comment this line# define USE_ ESP32_ SOFTWIRE to use the regular Wire. h library.Will cause ESP32 to fail compilation

 RP2040 connected to MPU6500,
Still only identifying the device, MPU6050 cannot output correct data
[image]

I have purchased a new MPU6500 module and will continue testing it tomorrow
Will your code support WING in the future? Will the barometer and GPS be added to provide automatic home function

Thank you!

qqqlab, 2023-12-15:
Pushed a new version today with some barometers.

Still fighting with I2C on the ESP32, added a note in hw_ESP32.h

Did not test on RP2040 today, hope it still works ;-)

> Will your code support WING in the future? Will the barometer and GPS be added to provide automatic home function
If possible, it would be better to support BMI 323 and ICM42688, as both BF and INAV now support these sensors. adding an SD card holder can record some data. If WIFI is utilized, the PID can be modified using web mode

These are certainly options. I'm working on barometers now, then maybe GPS. The rest probably won't happen (not enough time to work on this...) 

But if you feel like, please go ahead and add the new functions/sensors and send me a pull request.

> I am planning to design a 36X36 FC hardware

Sounds good! Actually RP2040 can do 6 hardware serial ports with the PIO see [here](https://github.com/qqqlab/madflight/blob/main/doc/Hardware%20Considerations.md)

01Space, 2023-12-16:
Thank you again! Attempting to connect to MPU6500 through I2C mode using RP2040 failed again, but when I connect to MPU6500 through SPI mode using RP2040, the measurement data of MPU6500 can be displayed.I have abandoned I2C mode.
I should enter the hardware design phase now, using RP2040+ELRS (SX1280+ESP8285)+MPU6500(or MPU9250,Actually, I prefer BMI 088 more )+BMP280(Actually, I prefer DPS310 more) +5VBEC
Can you give some advice on hardware design?
[image]

qqqlab, 2023-12-18:
Definitely use SPI for the IMU: faster and more stable. I would not bother to include ELRS on the board, but provide a 4 pin connector to connect an external ELRS module. Rather include a current shunt and voltage sense resistor divider (see new BAT_ADC pin on my pinout) or use an INA226. You can use more modern IMU/BARO chips, but you will need to write drivers for them as I'm focusing on easily available sensor breakout boards at the moment. Options like LCD or SDCARD will probably need SPI, which is already used for IMU or you need to sacrifice 4 pins for a second SPI. 

Looking forward to see your design!