qqqlab/madflight #13: Compilling last version 1.1.1 with an i2c imu doesn't work

GRacing69, 2024-05-19:
Hello

Firts of all, thanks for sharing rhis project! 😉

Everytime i compile your quadcopter.ino file example with any of the IMU_SPI , it compiles correctly , i export the compiled binary in uf2, and after loading it to the RP2040, it reeboots and i have acess to all the help and cli commands!
When i do a i2c (scan) the mpu6050 is detected at address 0X68!
But since i dont have any spi imu, only i2c, i defined use imu i2c mpu6050
It compiles witout errors , but after putting the uf2 exported compiled file in the rp2040, it loads it but the rp2040 doesnt reboot, no led flashes or blinks, and i loose the com port access to the rp2040, meaning that i cant do anything more!
No change in your original quadcopter ino file example except the IMU !
Any help is greatly appreciated

Regards

qqqlab, 2024-05-19:
Hi @GRacing69, thanks for giving madflight a try! 

I'm mainly developing on ESP32 currently, but quickly tried RP2040 v1.1.1 right now and could not get it to work at all. Unfortunately I currently don't have time to find out what is wrong. 

Please try [release v1.0.3](https://github.com/qqqlab/madflight/releases/tag/v1.0.3) for RP2040. I flight tested v1.0.3 with RP2040 + SPI MPU6500, with I2C MPU6050 your mileage may vary...

qqqlab, 2024-05-26:
@GRacing69 I've improved RP2040 support in v1.1.2-DEV: the IMU loop now runs on the second core, which keeps the first core free to respond to USB. I've tested it with an SPI IMU + CRSF remote. See also the updated readme: https://github.com/qqqlab/madflight?tab=readme-ov-file#gotchas

Maybe you can give it a try?

closed 2024-06-08 by qqqlab