qqqlab/madflight #17: Issue compiling library version 1.1.2 for esp32

Stillalive222, 2024-06-02:
`libraries\madflight\src\madflight/hw_ESP32/ESP32_PWM.h:65:20: error: 'ledcSetup' was not declared in this scope; did you mean 'ledc_stop'?
   65 |         act_freq = ledcSetup(ch, freq, bits);
      |                    ^~~~~~~~~
      |                    ledc_stop`
and
``libraries\madflight\src\madflight/hw_ESP32/ESP32_PWM.h:72:7: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   72 |       ledcAttachPin(pin, ch);
      |       ^~~~~~~~~~~~~
      |       ledcAttach

qqqlab, 2024-06-02:
Please try the current development version madflight v1.1.3-DEV (download directly from Github, not via Arduino IDE).

Or downgrade Arduino-ESP32 from v3.x.x to v2.x.x

Background: Arduino-ESP32 v3.0.0 was recently released, and has some breaking changes as you noticed. I was not able to fully test madflight v1.1.3 yet, it will be released later this week.

qqqlab, 2024-06-08:
madflight release 1.1.3 is compatible with Arduino-ESP32 v3.x.x

closed 2024-06-08 by qqqlab