qqqlab/madflight #18: Not install firmware from Arduino to ESP32 Nodemcu

skimans, 2024-06-12:
hi
i have problem to install firmware using Arduino on ESP32 NodeMcu.
When upload i obtain error message:

In file included from C:\Users\scdstf\Documents\Arduino\libraries\madflight\src/madflight.h:33,
                 from C:\Users\scdstf\Desktop\madflight-1.1.3\examples\01.Quadcopter\01.Quadcopter.ino:241:
C:\Users\scdstf\Desktop\madflight-1.1.3\examples\01.Quadcopter\01.Quadcopter.ino: In function 'void setup()':
C:\Users\scdstf\Documents\Arduino\libraries\madflight\src/madflight/hw_ESP32/hw_ESP32.h:14:42: error: expected ')' before 'ESP_ARDUINO_VERSION_STR'
 #define HW_ARDUINO_STR "Arduino-ESP32 v" ESP_ARDUINO_VERSION_STR
                                          ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\scdstf\Desktop\madflight-1.1.3\examples\01.Quadcopter\01.Quadcopter.ino:253:44: note: in expansion of macro 'HW_ARDUINO_STR'
     Serial.printf(MADFLIGHT_VERSION " on " HW_ARDUINO_STR " starting %d ...\n",i);
                                            ^~~~~~~~~~~~~~
C:\Users\scdstf\Desktop\madflight-1.1.3\examples\01.Quadcopter\01.Quadcopter.ino:253:18: note: to match this '('
     Serial.printf(MADFLIGHT_VERSION " on " HW_ARDUINO_STR " starting %d ...\n",i);
                  ^
exit status 1
Errore durante la compilazione per la scheda Node32s.

what can i do for solve ?

qqqlab, 2024-06-12:
Hmm, strange. Quick fix is to comment out the offending line 253 in Quadcopter.ino

skimans, 2024-06-12:
@qqqlab thanks much for info. Now work and i cna install on Esp32 nodeMCU.
I use MPU6050 connect I2C with SDA = 23 pin and SCL = 22 pin.
But now in Serial monitor i have this message:

IMU interrupt not fire

i have modify code in this way:

//========================================================================================================================//
//                                                 BOARD                                                                  //
//========================================================================================================================//
// Uncomment/change the following #include to the flight controller you want to use, or leave commented out to use the
// default board pinout (madflight_board_default_*.h). See library/madflight/src for all available boards

//#include <madflight_board_default_ESP32.h>

//========================================================================================================================//
//                                                 USER-SPECIFIED DEFINES                                                 //
//========================================================================================================================//

//--- RC RECEIVER
#define RCIN_USE  RCIN_USE_SBUS //RCIN_USE_CRSF, RCIN_USE_SBUS, RCIN_USE_DSM, RCIN_USE_PPM, RCIN_USE_PWM
#define RCIN_NUM_CHANNELS 6 //number of receiver channels (minimal 6)

//--- IMU SENSOR
#define IMU_USE  IMU_USE_I2C_MPU6050 // IMU_USE_SPI_BMI270, IMU_USE_SPI_MPU9250, IMU_USE_SPI_MPU6500, IMU_USE_SPI_MPU6000, IMU_USE_I2C_MPU9250, IMU_USE_I2C_MPU9150, IMU_USE_I2C_MPU6500, IMU_USE_I2C_MPU6050, IMU_USE_I2C_MPU6000
//Set sensor orientation. The label is yaw / roll (in that order) needed to rotate the sensor from it's normal position to it's mounted position.
//If not sure what is needed: use CLI 'proll' and try each setting until roll-right gives positive ahrs_roll, pitch-up gives positive ahrs_pitch, and yaw-right gives positive ahrs_yaw
#define IMU_ALIGN  IMU_ALIGN_CW0 //IMU_ALIGN_CW0, IMU_ALIGN_CW90, IMU_ALIGN_CW180, IMU_ALIGN_CW270, IMU_ALIGN_CW0FLIP, IMU_ALIGN_CW90FLIP, IMU_ALIGN_CW180FLIP, IMU_ALIGN_CW270FLIP
#define IMU_I2C_ADR  0x68 //IMU I2C address. If unknown, use CLI 'i2c'

I2C_chip_ scanner tell me this result:

===================
I2C Chip Scanner
===================
I2C: Scanning ...
I2C: Found address: 0x68 (decimal 104)
      MATCH: reg=0x75 result=0x68 -> MPU6000, MPU6050, or MPU9150 6/9DOF
I2C: Found 1 device(s)

skimans, 2024-06-13:
on serial monitor:

DEFAULT ESP32 BOARD - Espressif ESP32-DevKitC 38 pin
eeprom_read[64]:FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
CFG: EEPROM Config invalid, using defaults. len=65535 crc=FFFF crc_expected=10000
HW: MCU=ESP32 BOARD_NAME=DEFAULT ESP32 BOARD - Espressif ESP32-DevKitC 38 pin
HW_PIN: LED=2 SPI_MOSI=21 MISO=36 SCLK=19 IMU_CS=18 IMU_EXTI=39 I2C_SDA=23 SCL=22 OUT[11]=33,25,26,27,14,12,13,15,0,4,16 RCIN_RX=35 TX=32 GPS_RX=17 TX=5
I2C: Scanning ...
I2C: Found address: 0x68 (104)
I2C: Found 1 device(s)
RCIN_USE_SBUS
IMU: IMU_USE_I2C_MPU6500 sample_rate=1000Hz rv=0
IMU: IMU_EXEC_FREERTOS
BARO: BARO_USE_NONE
E (6117) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=5
E (6120) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=11
E (6132) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=21
E (6143) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=5
E (6154) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=11
E (6166) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=21
E (6177) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=5
E (6188) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=11
E (6200) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=21
E (6211) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=5
E (6222) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=11
E (6234) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=21
IMU interrupt not firing.
IMU interrupt not firing.
IMU interrupt not firing.
IMU interrupt not firing.

qqqlab, 2024-06-13:
you need to connect the MPU6050 interrupt pin IMU_EXTI=39

skimans, 2024-06-14:
thanks much @qqqlab. Now work fine and i have also calibrated IMU.

closed 2024-06-14 by skimans