qqqlab/madflight #43: Board rebooting if used ANY I2C IMU

tigrajack, 2025-03-16:
Board rebooting if used ANY I2C IMU such as IMU_USE_I2C_MPU6050.
This problem may have started with version 1.3.1

qqqlab, 2025-03-16:
Hi @tigrajack , thanks for reporting this. What processor are you using?

tigrajack, 2025-03-17:
> Hi [@tigrajack](https://github.com/tigrajack) , thanks for reporting this. What processor are you using?

ESP32 and ESP32S3. The problems look the same.

qqqlab, 2025-03-17:
Thanks for the compliments. It is a fun project to work on. I'm currently working on the next major release 2.0, so probably will not investigate this i2c issue for 1.3, but instead will make sure that it works on the new release.

I'm not much of a video guy, but it would be great if you could add something madflight related to youtube!

And: github stars are always welcome. Hint, hint :-)

tigrajack, 2025-03-18:
> Thanks for the compliments. It is a fun project to work on. I'm currently working on the next major release 2.0, so probably will not investigate this i2c issue for 1.3, but instead will make sure that it works on the new release.
For planning, when will be release 2.0?
> 
> I'm not much of a video guy, but it would be great if you could add something madflight related to youtube!
I'll think about it

tigrajack, 2025-03-18:
> Thanks for the compliments. It is a fun project to work on. I'm currently working on the next major release 2.0, so probably will not investigate this i2c issue for 1.3, but instead will make sure [...]

++

qqqlab, 2025-03-20:
Hi @tigrajack , thanks for the &star;

FYI: I just pushed an untested release 2.0, maybe you can download/clone it and give it a try? 

Some unfinished documentation is [here](https://github.com/qqqlab/madflight-docs/tree/main/docs)

saaraswata, 2025-03-20:
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379e6e
SPIWP:0xee
mode:DIO, clock div:1

I am going to fly plane tomorrow, any fix?

qqqlab, 2025-03-20:
@saaraswata Thanks for giving it a go. 

Should be fixed now in 2.0.0-DEV

saaraswata, 2025-03-21:
How to get access to 2.0.0-DEV? I am trying to make a video on flight controllers using ESP32, have tried ESPDRONE and ESPFC till now, currently I am trying to fly a plane and thus using your madflight.... It looks so much cleaner! It would be helpful if you could provide link to 2.0.0-DEV

qqqlab, 2025-03-21:
@saaraswata Please use the green Code button on [main github page](https://github.com/qqqlab/madflight), then select Download ZIP. Unpack the zip as subdirectory in your arduino/libraries directory. 

[image]

saaraswata, 2025-03-21:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac

nothing else shows up,  here is the config i used:

//This pin layout is optimized for Espressif ESP32-S3-DevKitC-1 (44 pin) board, use "ESP32-S3 Dev Module" as board in Arduino IDE

//see https://madflight.com for details

//ESP32-S3 - Most pins can be assigned freely

//Arduino "Serial" is uart0 (pin 1 tx, pin 3 rx) connected to serial->USB converter and is used for programming/CLI
//you could use Serial0/uart0 as third serial port, when using USB (not serial->USB) for programming/CLI

#pragma once

#define HAL_BOARD_NAME "ESP32-S3"

#define MADFLIGHT_BOARD R""(

// PINOUT

// Serial Pins
pin_ser0_rx    44 // Serial1/uart1
pin_ser0_tx    43 // Serial1/uart1
pin_ser0_inv   -1
pin_ser1_rx     3 // Serial2/uart2
pin_ser1_tx    46 // Serial2/uart2
pin_ser1_inv   -1
pin_ser2_rx    -1
pin_ser2_tx    -1
pin_ser2_inv   -1
pin_ser3_rx    -1
pin_ser3_tx    -1
pin_ser3_inv   -1
pin_ser4_rx    -1
pin_ser4_tx    -1
pin_ser4_inv   -1
pin_ser5_rx    -1
pin_ser5_tx    -1
pin_ser5_inv   -1
pin_ser6_rx    -1
pin_ser6_tx    -1
pin_ser6_inv   -1
pin_ser7_rx    -1
pin_ser7_tx    -1
pin_ser7_inv   -1

// SPI Pins
pin_spi0_miso  12
pin_spi0_mosi  11
pin_spi0_sclk  13
pin_spi1_miso  -1
pin_spi1_mosi  -1
pin_spi1_sclk  -1
pin_spi2_miso  -1
pin_spi2_mosi  -1
pin_spi2_sclk  -1
pin_spi3_miso  -1
pin_spi3_mosi  -1
pin_spi3_sclk  -1

// I2C Pins
pin_i2c0_sda    8
pin_i2c0_scl    9
pin_i2c1_sda   14
pin_i2c1_scl   13
pin_i2c2_sda   -1
pin_i2c2_scl   -1
pin_i2c3_sda   -1
pin_i2c3_scl   -1

// OUT Pins
pin_out0        4
pin_out1        5
pin_out2        6
pin_out3        7
pin_out4       15
pin_out5       16
pin_out6       -1
pin_out7       -1
pin_out8       -1
pin_out9       -1
pin_out10      -1
pin_out11      -1
pin_out12      -1
pin_out13      -1
pin_out14      -1
pin_out15      -1

// Other Pins
pin_bat_i      -1
pin_bat_v      -1
pin_bbx_cs     -1
pin_imu_cs     10
pin_imu_int    14
pin_led         2
led_on      HIGH_IS_ON
pin_mmc_dat    40
pin_mmc_clk    39
pin_mmc_cmd    38
pin_rcl_ppm    -1

// BUSSES

// Serial Busses
rcl_ser_bus     1
gps_ser_bus     2
rdr_ser_bus    -1

// SPI Busses
imu_spi_bus    -1  //setup in .ino
bbx_spi_bus    -1

// I2C Busses
imu_i2c_bus    -1 //setup in .ino
bar_i2c_bus     0
mag_i2c_bus     0
bat_i2c_bus     0

)""
//end of #define MADFLIGHT_BOARD

qqqlab, 2025-03-21:
@saaraswata 

I just updated the plane example, your board dump still shows an older version.

You should get at least some "madflight starting" messages before things crash. Are you sure you have your board correctly configured in the Arduino IDE? The ESP32-S3 is a bit tricky to get configured. Use a "hello world" sketch to output something on Serial to confirm.

saaraswata, 2025-03-21:
Thanks, I noticed after downloading that the plane example was left, I am downloading again

saaraswata, 2025-03-21:
In file included from c:\Users\Dell\Documents\Arduino\libraries\madflight\src/madflight_board.h:4,
                 from C:\Users\Dell\AppData\Local\Temp\.arduinoIDE-unsaved2025221-2920-zznz8m.w3h6l\03.Plane\madflight_config.h:17,
                 from C:\Users\Dell\AppData\Local\Temp\.arduinoIDE-unsaved2025221-2920-zznz8m.w3h6l\03.Plane\03.Plane.ino:76:
c:\Users\Dell\Documents\Arduino\libraries\madflight\src/madflight_board__ESP32-S3.h:14:25: error: unterminated raw string
   14 | #define MADFLIGHT_BOARD R""(
      |                         ^
exit status 1

Compilation error: exit status 1

Many other places have this same thing too, fix this please

saaraswata, 2025-03-21:
In file included from c:\Users\Dell\Documents\Arduino\libraries\madflight\src/madflight_board.h:4,
                 from C:\Users\Dell\AppData\Local\Temp\.arduinoIDE-unsaved2025221-2920-zznz8m.w3h6l\03.Plane\madflight_config.h:17,
                 from C:\Users\Dell\AppData\Local\Temp\.arduinoIDE-unsaved2025221-2920-zznz8m.w3h6l\03.Plane\03.Plane.ino:76:
c:\Users\Dell\Documents\Arduino\libraries\madflight\src/madflight_board__ESP32-S3.h:19:1: error: 'pin_ser0_rx' does not name a type
   19 | pin_ser0_rx    18 // Serial1/uart1
      | ^~~~~~~~~~~
In file included from C:\Users\Dell\AppData\Local\Temp\.arduinoIDE-unsaved2025221-2920-zznz8m.w3h6l\03.Plane\03.Plane.ino:77:
c:\Users\Dell\Documents\Arduino\libraries\madflight\src/madflight.h: In function 'void madflight_setup()':
c:\Users\Dell\Documents\Arduino\libraries\madflight\src/madflight.h:104:6: error: 'madflight_config' was not declared in this scope; did you mean 'madflight_die'?
  104 |   if(madflight_config) {
      |      ^~~~~~~~~~~~~~~~
      |      madflight_die

exit status 1

Compilation error: exit status 1

qqqlab, 2025-03-21:
@saaraswata 

It looks to me that you did not fully replace the madflight lib. Delete the libraries/madflight folder and replace it with a fresh unzipped github zip.

saaraswata, 2025-03-21:
I have downloaded from github the latest one, updated 3 hours ago, still it has this issue

In file included from C:\Users\maamo\AppData\Local\Temp\.arduinoIDE-unsaved2025221-7292-1xk7ekj.vnt1\03.Plane\03.Plane.ino:76:
C:\Users\maamo\AppData\Local\Temp\.arduinoIDE-unsaved2025221-7292-1xk7ekj.vnt1\03.Plane\madflight_config.h:28:26: error: unterminated raw string
   28 | #define MADFLIGHT_CONFIG R""(
      |                          ^
C:\Users\maamo\AppData\Local\Temp\.arduinoIDE-unsaved2025221-7292-1xk7ekj.vnt1\03.Plane\madflight_config.h:75:25: error: unterminated raw string
   75 | #define MADFLIGHT_BOARD R""(
      |                         ^
exit status 1

Compilation error: unterminated raw string

saaraswata, 2025-03-22:
Okay, i have installed 1.2.2 and it works fine, had to decrease task priority to 24 for freertos.... also i modified the ppm code for radiolink receiver 8 channel which im using... btw sbus in latest one has some errors, when i tried compiling with sbus it showed some errors. in 1.2.2 sbus compiles fine but it doesnt read from my radiolink receiver can you check that?

qqqlab, 2025-03-22:
@saaraswata good new that you have it working now, but a pity that you had to go back intime to v1.2.2

I can't check sbus as I don't have it, but it would be nice if you could get it to work. If i remember correctly, sbus uses inverted serial, maybe that is the problem.

qqqlab, 2025-03-23:
@saaraswata 

What are you using to compile madflight? Arduino IDE or PlatformIO?

Reason for asking is that I don't like the idea that v2.0.0 does not compile ...

saaraswata, 2025-03-23:
I'm using arduino IDE, v2.x.x I guess the latest release, also using latest esp32 sdk

qqqlab, 2025-03-23:
@saaraswata thanks!

I borrowed a SBUS receiver, and got it to work in 2.0.0. Maybe you could give it try?

And, if you installed a fresh v2 copy, could you please open the new HelloWorld example and compile it without making any changes? I would like to make sure it compiles now.

saaraswata, 2025-03-24:
Umm I got another problem, my plane crashed.... I changed flight mode
midflight in 1.2.2 and throttle response stopped. If I recall correctly
there main code for handling pid is still the same? The plane was
oscillating midflight so had to switch to manual mode and boom! Throttle
stopped, Also there was a Bug in FBWA mode, if I take it to full throttle,
pmot command shows that motor output is 54 percent, never goes above it.
And it it stays at 54 for 5 seconds.of so it automatically says rcin
disconnected unarmed even though signal is there

saaraswata, 2025-03-24:
> [@saaraswata](https://github.com/saaraswata) thanks!
>
> I borrowed a SBUS receiver, and got it to work in 2.0.0. Maybe you could give it try?
>
> And, if you installed a fresh v2 copy, could you please [...]

Okay I will try it to compile and flash today

saaraswata, 2025-03-24:
> @saaraswata <https://github.com/saaraswata>
>
> Umm I got another problem, my plane crashed....
>
> Sorry for that, I hope you can still glue it back together. Then make sure
> that you can switch to manual mode. Not a big help for me though - I
> usually crash when flying manual ;-) And, get logging to work, that way you
> can do at least some post-mortem research.

Thanks for the tip, BTW I'm using esp32 which has 16MB of flash, so is it
possible to use part of it for logging?

qqqlab, 2025-03-24:
@saaraswata 

That was also my initial thought: to use the internal flash for logging. However, it messes things up majorly, as the program code is running from the same flash....

So, I went for SD card. You could try this near zero cost solution:

[Connect a microSD card using a microSD to SD adapter](https://alexlubbock.com/micro-sd-adapter-esp8266-esp32)

saaraswata, 2025-03-24:
Actually previously I have used esp fc firmware for drones and it allowed
logging to sd card, I didn't see any losses though

tigrajack, 2025-03-24:
[env:madflight_esp32]
platform = espressif32
board = esp32dev
framework = arduino

Is not work for  Arduino-ESP32 v3.x.x

How to set up PlatformIO to work with Arduino-ESP32 v3.x.x?

qqqlab, 2025-03-24:
@tigrajack 

No idea, sorry. You'll have to search for an answer, I mainly use the Arduino IDE...

tigrajack, 2025-03-24:
> [@tigrajack](https://github.com/tigrajack)
> 
> No idea, sorry. You'll have to search for an answer, I mainly use the Arduino IDE...

https://github.com/platformio/platform-espressif32/issues/1225
PlatformIO not supported for Arduino 3.x ((

tigrajack, 2025-03-25:
I see you've gone back to 1.3.4. 
When do you plan to move to 2.0.0 or fix the reboot I2C in 1.3.4?

tigrajack, 2025-03-25:
> [@tigrajack](https://github.com/tigrajack)
> 
> Thanks for the update on PlatformIO, even though it is not good news... Maybe madflight still compiles with Arduino 2.x?
YES it is work for Arduino 2.x!
if little fix madflight\hal\ESP32\ESP32_PWM_v3.cpp
               madflight\hal\ESP32\ESP32_PWM_v3.h
AND fix platformio.ini
[env:esp32s3]
platform = espressif32
framework = arduino
board = lolin_s3_mini
monitor_speed = 115200
build_flags =
     -DCONFIG_IDF_TARGET_ESP32S3
    -DARDUINO_USB_MODE=1
    -DARDUINO_USB_CDC_ON_BOOT=1

qqqlab, 2025-04-01:
@saaraswata @tigrajack 

I just released v2. 

I'm closing this issue, as the stated issue "esp rebooting on i2c sensor" does not occur in v2.

Many thanks for your feedback, and feel free to open a new Issue or Discussion !

closed 2025-04-01 by qqqlab