qqqlab/madflight #46: Git clone of repo in Win fails

geosmall, 2025-04-12:
Tried git clone in Windows, I get the following error:

```
$ git clone https://github.com/qqqlab/madflight.git
Cloning into 'madflight'...
remote: Enumerating objects: 9906, done.
remote: Counting objects: 100% (1107/1107), done.
remote: Compressing objects: 100% (417/417), done.
remote: Total 9906 (delta 855), reused 847 (delta 683), pack-reused 8799 (from 1)
Receiving objects: 100% (9906/9906), 19.67 MiB | 20.70 MiB/s, done.
Resolving deltas: 100% (7965/7965), done.
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'src/madflight/ahr/Madgwick/Madgwick.h'
  'src/madflight/ahr/madgwick/madgwick.h'
  'src/madflight/ahr/Mahony/Mahony.h'
  'src/madflight/ahr/mahony/mahony.h'
```

qqqlab, 2025-04-12:
Thanks for reporting this. Should be fixed now, I've deleted the old (lowercase) directories.

If you like madflight, please give it a ⭐

geosmall, 2025-04-15:
Thanks! This fixes the issue.  On to compiling with Arduino IDE.  Feel free to close this issue.

closed 2025-04-15 by geosmall