qqqlab/madflight #47: Compile in Arduino IDE on Win11 fails

geosmall, 2025-04-15:
Compiling for STM32F411 Black Pill F411CE on Win11 / Arduino IDE 1.8.19 fails with:

`
In file included from D:\Arduino\libraries\madflight\src/madflight_board.h:19,
                 from D:\Arduino\libraries\madflight\examples\00.HelloWorld\madflight_config.h:29,
                 from D:\Arduino\libraries\madflight\examples\00.HelloWorld\00.HelloWorld.ino:13:
D:\Arduino\libraries\madflight\src/madflight_board_STM32.h:50:25: error: unterminated raw string
   50 | #define MADFLIGHT_BOARD R""(
      |                         ^
madflight_config.h:31:26: error: unterminated raw string
   31 | #define MADFLIGHT_CONFIG R""(
      |                          ^
`

geosmall, 2025-04-15:
Maybe something about Linux lf vs Win crlf?

A little bit of digging is telling me that raw string literals do support multiple lines if they’re written as one continuous token in the compiler’s view. But when combined with a #define macro, the newline in the middle can lead the preprocessor to assume the literal is “unterminated.” It expects everything from R"( to )" to be on one logical line, or at least parsed as a single token.

The C++ Standard only guarantees multiline raw strings if they remain in a single token in the normal compiler sense — macros can introduce complications with newlines through the preprocessor.  I'm wondering if putting the raw string in a normal variable Instead of a macro would be more portable?

geosmall, 2025-04-15:
To keep it all in one header file while still using a raw string literal for definitions, one could use an inline variable (a C++17 feature) rather than a macro.

This way, multiple source files can include the same header without causing multiple-definition errors, and you avoid multiline macro pitfalls.  But it does require C++17 or newer.

qqqlab, 2025-04-15:
@geosmall Thanks for digging into this. 

I never had issues compiling with the Windows Arduino IDE, but had others complain about this before. 

I was hoping that I found a clean-and-easy way to include a multi-line config - too bad that this does not work...

I'll probably go with 
```c
const char* MADFLIGHT_BOARD = R""(
// multiline text
)"";

...

const char* MADFLIGHT_CONFIG = R""(
// multiline text
)"";
```
but in the madflight_config.h and madflight_board_xxx.h header files. As these headers should be included only once in the main program, it should not cause issues.

Unless you have a better idea 🤔 

Sticking it in a .ccp file won't work with the Arduino IDE, as it compiles *all* cpp files it finds.

qqqlab, 2025-04-15:
The issue is apparently the Arduino IDE version:

Arduino IDE version 2.3.6 works fine, version 1.8.19 doesn't.

geosmall, 2025-04-16:
I believe the cause of this issue is a non-portable macro implementation that may or may not compile.  Feel free to close this issue at your discretion.

qqqlab, 2025-04-16:
BTW @geosmall , why are you still using Arduino IDE v1 ? I switched to v2 a while back, I had some hick-ups with the early v2 releases, but now it is running smoothly for me. Just wondering why apparently so many people are still using v1...

geosmall, 2025-04-17:
Thanks for the fix.  All examples compiled successfully for STM32F411 Black Pill F411CE on Win11 / Arduino IDE 1.8.19 / Arduino_Core_STM32 v2.7.1. I think we can consider this issue to be addressed.

> BTW [@geosmall](https://github.com/geosmall) , why are you still using Arduino IDE v1 ? I switched to v2 a while back, I had some hick-ups with the early v2 releases, but now it is running smoothly for me. Just wondering why apparently so many people are still using v1...

My workflow for Arduino type projects is to use an external editor (Sublime Text / VS Code mix), Segger O-Zone with J-Link for debugger and the v1 IDE (as a portable install) to compile.  I tried v2 IDE a while back and really did not like it.  Bloated and slow to load/close.  Lots of bugs made it unreliable.  Sort of like the early versions of v1 so not surprising really.  IDE v1 really didn't reaxh stable until v1.6.x.  This does leave me on Arduino_Core_STM32 v2.7.1 but since I mostly use it for the gcc compiler not a big problem for me yet.

I may have to try v2 again to see how it has progressed, but do not have a driving reason to do so.

qqqlab, 2025-04-17:
@geosmall Thanks for the IDE v1 explanation. Actually the problem was not IDE v1 in general, as I was able to compile for RP2350 on it. Leaves Arduino_Core_STM32 v2.7.1 as the probably cause, but I did not investigate further as it works now.

closed 2025-04-17 by qqqlab