qqqlab/madflight #48: Radio controller not working

asGenn, 2025-05-02:
Hello,
I'm using the FlySky i6 transmitter with the FS-iA6B receiver. This receiver supports either iBUS or PPM signal output. Since your code doesn't include support for iBUS, I decided to use PPM instead. However, I'm encountering a problem:

When I connect the receiver's RX (PPM output) pin to any GPIO pin on my ESP32 Devkit v1 board, I don't receive any signal. For example, when I try to run `prcl` or `calradio`, I get the following error:

```
==== Radio Calibration ====
ERROR: radio not connected. Connect and try again

==== Radio Calibration Aborted ====
```

I have already enabled the PPM output setting from the transmitter, so that should be correct.

My question is: Am I wiring something incorrectly, or is there a different approach I need to take to get the ESP32 to read PPM signals properly from the FS-iA6B receiver?

[image]

[image]

[image]

qqqlab, 2025-05-02:
Hi @asGenn , thanks for trying madflight.

Did you set the following in madflight_config?

```
rcl_gizmo PPM
pin_rcl_ppm <your ppm pin>
```

If yes: are you absolutely sure your receiver is outputting PPM?

asGenn, 2025-05-02:
Hi again @qqqlab ,
I enabled **PPM output** on my FlySky i6 transmitter .

[image]

However, I'm not entirely sure if the receiver is actually outputting PPM correctly. On the transmitter, I’ve activated PPM mode under “RX Setup > Output Mode,” and I connected the signal wire from **channel 1 (CH1)** of the FS-iA6B (which is the correct PPM output port) to one of the ESP32's GPIO pins.

the configuration file:

```
/*========================================================================================================================
                                              MADFLIGHT CONFIG
==========================================================================================================================

The configuration is loaded from two strings: first from madflight_board in <madflight_board.h>, and then from 
madflight_config in this file. If the same setting occurs more than once, the last one is applied.

The strings are multi-line raw strings in with a key-value list. Anything after '#' or '/' is ignored as comment.

You have 3 options to setup the flight controller:

  1) Default - Keep "#include <madflight_board.h>", this defines the default pinout as shown on https://madflight.com 
     for the supported processor families. Now edit madflight_config below, and uncomment and configure imu_gizmo, 
     imu_bus_type, rcl_gizmo, and other lines as needed. (Do not change the pin_xxx_yyy and xxx_yyy_bus settings.)
 
  2) BetaFlight - Change "#include <madflight_board.h>" to the BetaFlight flight controller you want to use, for example: 
     "#include <madflight_zzz_MTKS-MATEKH743.h>". See library/madflight/src for all available boards. Edit madflight_config
     to fine-tune the configuration.
 
  4) Bare Metal - Remove "#include <madflight_board.h>", and set the full configuration in madflight_config below.

Pins and spi/i2c/serial busses use zero-based numbering, i.e. "gps_ser_bus 0" connects the GPS to the first serial bus
with pins pin_ser0_tx and pin_ser0_rx. Pins use GPIO numbers, not physical pin numbers. Use -1 to disable a pin or bus.

You can also modify the configuration from the CLI, for example "set imu_gizmo MPU6500" or "set imu_spi_bus 1", then
use "cwrite" to save the config to eeprom and reboot to use the new config.

If things do not work as expected, have a good look at the startup messages!

========================================================================================================================*/

#include <madflight_board.h>

const char madflight_config[] = R""(

//--- IMU --- Inertial Measurement Unit  (use spi -OR- i2c bus)
//imu_gizmo      MPU6050    // options: NONE, BMI270, MPU6000, MPU6050, MPU6500, MPU9150, MPU9250, ICM45686, ICM42688
//imu_bus_type   I2C     // options: SPI, I2C (not all combinations of gizmo and bus_type are supported)
//imu_align      CW0     // options: CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP
//imu_spi_bus    -1 //spi
//pin_imu_cs     -1 //spi
//pin_imu_int    19 //spi and i2c
//imu_i2c_bus    0 //i2c
//imu_i2c_adr     104 //i2c: enter decimal i2c address, not hex (use 0 for default i2c address)

// IMPORTANT: the IMU sensor should be the ONLY sensor on the selected bus

//--- RCL --- Remote Controller Link  (use serial bus -OR- ppm pin)
rcl_gizmo      PPM  // options: NONE, MAVLINK, CRSF, SBUS, DSM, PPM
rcl_num_ch     6     // number of channels
rcl_deadband   20     // center stick deadband
//rcl_ser_bus   -1
pin_rcl_ppm   27

//--- BAR --- Barometer
//bar_gizmo      NONE  // options: NONE, BMP390, BMP388, BMP280, MS5611, HP203B
//bar_i2c_adr    0
//bar_i2c_bus   -1

//--- MAG --- Magnetometer
[... 74 lines cut ...]
//pin_out9      -1
//pin_out10     -1
//pin_out11     -1
//pin_out12     -1
//pin_out13     -1
//pin_out14     -1
//pin_out15     -1

)""; // End of madflight_config

//========================================================================================================================//
//                                               COMPILER OPTIONS                                                         //
//========================================================================================================================//

// Reset config eeprom to defaults (uncomment this, upload, execute, then comment out, and upload again)
//#define MF_CONFIG_CLEAR

// Uncomment to print additional debug information and reduce startup delay
//#define MF_DEBUG

```

I'm not sure which GPIO pin on the ESP32 I should use here. Does the ESP32 have a specific pin for receiving PPM input? Or can I use any digital GPIO input (like GPIO 4 or 15)? Also, is there any way to confirm that the PPM signal is actually being transmitted from the receiver?

I would appreciate it if you could clarify how to properly wire and configure PPM input with an ESP32. Thanks!

qqqlab, 2025-05-05:
As this appears to me an issue with your setup and not with madflight, I'm moving this to the Discussion section.

On the ESP32 you can use almost any pin for PPM input. Please use an oscilloscope, digital analyzer, or a small ESP32 program to check that your receiver is outputting a valid PPM signal. You'll find enough resources online on both ESP32 and PPM.