qqqlab/madflight #54: v2.0.1 linking fails on VsCode

vidma, 2025-06-07:
Been trying to convert Plane example to the new code structure (with goal to later try adding VTAIL output mixer or something alike - no plans?), however got stuck with these linking issues... seems compile but madflight code not linkable - e.g. `undefined reference to `cfg'`.
 
any clue?  bug? CPP standard, toolchain (it seems I've update all VSCode extentions/dependencies)?

P.S. nothing changes if I use `board = rpipico2` in `platform.io` 

```
Processing madflight_pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pimoroni_tiny2350; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pimoroni_tiny2350.html
PLATFORM: Raspberry Pi RP2040 (1.17.0+sha.e89b79e) > Tiny2350
HARDWARE: RP2350 150MHz, 512KB RAM, 4MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40504.0+sha.948e9bf 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
Converting main.ino
Flash size: 4.00MB
Sketch size: 4.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 4190208 EEPROM start: 0x103ff000 Filesystem start: 0x103ff000 Filesystem end: 0x103ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 74 compatible libraries
Scanning dependencies...
Dependency Graph
|-- src
Building in release mode
Compiling .pio/build/madflight_pico/src/main.ino.cpp.o
Linking .pio/build/madflight_pico/firmware.elf
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `_ZN16ImuGizmoICM426XX13getMotion6NEDEPfS0_S0_S0_S0_S0_':
main.ino.cpp:(.text._ZN16ImuGizmoICM426XX13getMotion6NEDEPfS0_S0_S0_S0_S0_[_ZN16ImuGizmoICM426XX13getMotion6NEDEPfS0_S0_S0_S0_S0_]+0x10): undefined reference to `_ZN8ICM426XX4readEPs'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `_ZN16ImuGizmoICM4568613getMotion6NEDEPfS0_S0_S0_S0_S0_':
main.ino.cpp:(.text._ZN16ImuGizmoICM4568613getMotion6NEDEPfS0_S0_S0_S0_S0_[_ZN16ImuGizmoICM4568613getMotion6NEDEPfS0_S0_S0_S0_S0_]+0x12): undefined reference to `_ZN8ICM456xx15getDataFromFifoER19inv_imu_fifo_data_t'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `_ZN16ImuGizmoICM426XX6createEP9ImuConfigP8ImuState':
main.ino.cpp:(.text._ZN16ImuGizmoICM426XX6createEP9ImuConfigP8ImuState[_ZN16ImuGizmoICM426XX6createEP9ImuConfigP8ImuState]+0x26): undefined reference to `_ZN8ICM426XX6detectEP13MPU_Interface'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._ZN16ImuGizmoICM426XX6createEP9ImuConfigP8ImuState[_ZN16ImuGizmoICM426XX6createEP9ImuConfigP8ImuState]+0x84): undefined reference to `_ZN8ICM426XX9type_nameEv'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `_ZN3Imu6updateEv':
main.ino.cpp:(.text._ZN3Imu6updateEv+0x268): undefined reference to `cfg'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `_ZN3Imu5setupEv':
main.ino.cpp:(.text._ZN3Imu5setupEv+0xa): undefined reference to `_ZN8CfgClass11printModuleEPKc'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._ZN3Imu5setupEv+0x17e): undefined reference to `_ZN8ICM456xxC1ERN7arduino11HardwareSPIEhm'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._ZN3Imu5setupEv+0x304): undefined reference to `cfg'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `_Z13madflight_dieN7arduino6StringE':
main.ino.cpp:(.text._Z13madflight_dieN7arduino6StringE+0x36): undefined reference to `_ZN3Led6toggleEv'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z13madflight_dieN7arduino6StringE+0x52): undefined reference to `_ZN3Cli6updateEv'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z13madflight_dieN7arduino6StringE+0x5e): undefined reference to `_ZN3Rcl6updateEv'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z13madflight_dieN7arduino6StringE+0x64): undefined reference to `led'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z13madflight_dieN7arduino6StringE+0x74): undefined reference to `cli'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z13madflight_dieN7arduino6StringE+0x78): undefined reference to `rcl'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `_Z14madflight_warnN7arduino6StringE':
main.ino.cpp:(.text._Z14madflight_warnN7arduino6StringE+0x2c): undefined reference to `_ZN3Led6toggleEv'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z14madflight_warnN7arduino6StringE+0x4c): undefined reference to `led'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `loop':
main.ino.cpp:(.text.loop+0x4): undefined reference to `_ZN3Mag6updateEv'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text.loop+0x28): undefined reference to `mag'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text.loop+0x30): undefined reference to `cli'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text.loop+0x22): undefined reference to `_ZN3Cli6updateEv'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: .pio/build/madflight_pico/src/main.ino.cpp.o: in function `_Z9led_Blinkv':
main.ino.cpp:(.text._Z9led_Blinkv+0x1c): undefined reference to `_ZN3Led3setEb'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z9led_Blinkv+0x40): undefined reference to `out'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z9led_Blinkv+0x44): undefined reference to `led'
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text._Z9led_Blinkv+0x36): undefined reference to `_ZN3Led3setEb'
[... cut ...]
/Users/vidma/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: main.ino.cpp:(.text.setup+0x50): undefined reference to `out'
collect2: error: ld returned 1 exit status
*** [.pio/build/madflight_pico/firmware.elf] Error 1
============================================================================================== [FAILED] Took 3.04 seconds ==============================================================================================

 *  The terminal process "platformio 'run', '--environment', 'madflight_pico'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

qqqlab, 2025-06-07:
No idea, I'm not an PlatformIO expert, I use the Arduino IDE.

If you find a solution, please post it and I'll update the docs.

vidma, 2025-06-08:
this is weird, however, what seem to have helped is using a git repo directly, instead of a directory for `madflight` lib:

```ini
[env]
;lib_extra_dirs = /Users/vidma/flyuav/px4_rpi/madflight/madflight

lib_deps =
    https://github.com/qqqlab/madflight.git

[env:pimoroni_tiny2350]
#platform = raspberrypi
platform = https://github.com/maxgerhardt/platform-raspberrypi.git

board = pimoroni_tiny2350
framework = arduino
board_build.core = earlephilhower
```

FYI, I still had to comment this line out: 

https://github.com/qqqlab/madflight/blob/8b41cbc48b6a532974b0cadf208bccbabac25b20/src/madflight/alt/alt_cpp.h#L57

and FYI in `Plane.ino.cpp` file, define the forward reference to some functions - because of the way they are ordered in code (dunno maybe there's a compiler flag for this instead):
```cpp
void led_Blink();
void control_FBWA(bool zero_integrators);
void control_ROLL(bool zero_integrators);
void out_KillSwitchAndFailsafe();
void out_Mixer();
void control_MANUAL();
```

now plenty of madflight .c/.cpp files are compiled and linking succeeds:
```
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/ahr/Madgwick/Madgwick.cpp.o
Archiving .pio/build/pimoroni_tiny2350/lib44d/libSD.a
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/ahr/Mahony/Mahony.cpp.o
Indexing .pio/build/pimoroni_tiny2350/lib44d/libSD.a
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/ahr/ahr.cpp.o
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/ahr/vqf/vqf.c.o
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/alt/alt_kalman2/altitude_kf.cpp.o
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/alt/alt_kalman3/KalmanFilter.cpp.o
...
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/gps/gps.cpp.o
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/imu/ICM426XX/ICM426XX.cpp.o
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/imu/ICM45686/ICM45686.cpp.o
Compiling .pio/build/pimoroni_tiny2350/lib57d/madflight/madflight/imu/ICM45686/imu/inv_imu_driver.c.o
...
Linking .pio/build/pimoroni_tiny2350/firmware.elf
Generating UF2 image
picotool uf2 convert -t elf ".pio/build/pimoroni_tiny2350/firmware.elf" ".pio/build/pimoroni_tiny2350/firmware.uf2"
Retrieving maximum program size .pio/build/pimoroni_tiny2350/firmware.elf
Flash size: 4.00MB
Sketch size: 4.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 4190208 EEPROM start: 0x103ff000 Filesystem start: 0x103ff000 Filesystem end: 0x103ff000
Checking size .pio/build/pimoroni_tiny2350/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.4% (used 33712 bytes from 524288 bytes)
Flash: [=         ]   6.0% (used 250516 bytes from 4190208 bytes)
Building .pio/build/pimoroni_tiny2350/firmware.bin
Building .pio/build/pimoroni_tiny2350/firmware.bin.signed
======================================================================================= [SUCCESS] Took 13.19 seconds =======================================================================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

vidma, 2025-06-08:
FYI importing the Arduino project didn't help, using `lib_deps` from github helped

qqqlab, 2025-06-08:
Thanks. I'm moving this to discussions, maybe it will help somebody else. (Supporting PlatformIO is not a priority for me.)