qqqlab/madflight #78: Compilation error on STM32F7 from hal/MF_Serial

EnzoPB, 2025-09-13:
Hi,
First of all thank you for this project. It looks awesome!
I'm trying to flash the basic "HelloWorld" example on an STM32F722-based commercial flight controller (the [DFR F722 Dual HD](https://www.drone-fpv-racer.com/en/dfr-f722-dual-hd-6814.html)), but it fails with the following error:

```
In file included from /home/enzo/Arduino/libraries/madflight/src/cli/../rcl/rcl.h:27,
                 from /home/enzo/Arduino/libraries/madflight/src/cli/../rcl/RclGizmoMavlink.h:41,
                 from /home/enzo/Arduino/libraries/madflight/src/cli/cli.h:28,
                 from /home/enzo/Arduino/libraries/madflight/src/cli/cli.cpp:1:
/home/enzo/Arduino/libraries/madflight/src/cli/../rcl/../hal/MF_Serial.h: In instantiation of 'int MF_SerialPtrWrapper<T>::read(uint8_t*, int) [with T = USBSerial*; uint8_t = unsigned char]':
/home/enzo/Arduino/libraries/madflight/src/cli/../rcl/../hal/MF_Serial.h:35:9:   required from here
   35 |     int read(uint8_t *buf, int len)  override { return _serial->readBytes(buf, len); }
      |         ^~~~
/home/enzo/Arduino/libraries/madflight/src/cli/../rcl/../hal/MF_Serial.h:35:75: error: invalid conversion from 'uint8_t*' {aka 'unsigned char*'} to 'char*' [-fpermissive]
   35 |     int read(uint8_t *buf, int len)  override { return _serial->readBytes(buf, len); }
      |                                                                           ^~~
      |                                                                           |
      |                                                                           uint8_t* {aka unsigned char*}
In file included from /home/enzo/.arduino15/packages/STMicroelectronics/hardware/stm32/2.11.0/cores/arduino/WSerial.h:7,
                 from /home/enzo/.arduino15/packages/STMicroelectronics/hardware/stm32/2.11.0/cores/arduino/wiring.h:48,
                 from /home/enzo/.arduino15/packages/STMicroelectronics/hardware/stm32/2.11.0/cores/arduino/Arduino.h:36,
                 from /home/enzo/Arduino/libraries/madflight/src/cli/cli.h:27:
/home/enzo/.arduino15/packages/STMicroelectronics/hardware/stm32/2.11.0/cores/arduino/../../libraries/USBDevice/inc/USBSerial.h:39:36: note:   initializing argument 1 of 'virtual size_t USBSerial::readBytes(char*, size_t)'
   39 |     virtual size_t readBytes(char *buffer, size_t length);  // read chars from stream into buffer
      |                              ~~~~~~^~~~~~
exit status 1

Compilation error: exit status 1
```

The code I'm compiling is the untouched Hello World example, only with the board config loaded with `#define MF_BOARD "brd/betaflight/DFRA-DFR_F722_DUAL_HD.h"`.

For now I've managed to find a workaround by casting `buf` to a `char*` in hal/MF_Serial.h: `return _serial->readBytes((char*) buf, len);`

The only other board I have to test this is another flight controller with a STM32F303. It is not officially supported by MF, and the compiled program ends up being too big to upload, but it seems to compile fine since it reaches the program size check.

Let me know if you need any additional information

EnzoPB, 2025-09-14:
I am using 

- Arduino IDE 2.3.6
- madflight 2.1.2
- stm32duino 2.11.0
- OS Debian 14 Linux 6.16.3+deb14-amd64

These are my compiler settings:

- Board Generic: STM32F7 series
- ...
- Board part number: Generic F722RETx
- C Runtime library: Newlib Nano + Float Printf
- Upload method: STM32CubeProgrammer (DFU)
- USB support: CDC (generic 'Serial' supersede U(S)ART)
- U(S)ART support: Enabled (generic 'Serial')

I tried playing around with these settings, and when I set "USB Support" to "CDC (No generic Serial)", it does compile, but nothing appears on the Serial, and the board does seems to react to incoming serial commands.

qqqlab, 2025-09-15:
Thanks, I think I understand what the issue is. It helps if you actually read the error message ;-)

In cli.cpp a MF_Serial wrapper is created for `Serial`, which can be a USBSerial depending on the IDE settings.

Problem is that USBSerial does not have the uint8_t* variant of the ReadBytes method, but HardwareSerial has.

So, your fix is fine for STM32. If the fix also works for the other platforms I'll include in the next madflight release.