qqqlab/madflight #79: ICM42688P interrupt not firing

ejowerks, 2025-09-16:
Awesome project and code, thank you for all your hard work! :-)

Testing your Hello World Example on a SPBE-SPEEDYBEEF405MINI, only change is the board define line in madflight_config.h (and removing the "P" from the imu_gizmo ICM42688 line in brd/betaflight/SPBE-SPEEDYBEEF405MINI.h). Madflight now detects the sensor but the interrupt isn't firing.

```
//define the board header to use, or comment out for none
//#define MF_BOARD "brd/default.h"
#define MF_BOARD "brd/betaflight/SPBE-SPEEDYBEEF405MINI.h"
```

Arduino IDE v2.3.6
stmduino v2.11.0
madflight v2.1.2
windoze 11

```
Arduino library: Arduino_Core_STM32 v(0x02U).(0x0BU).(0x00U)
Board: BETAFLIGHT-SPBE-SPEEDYBEEF405MINI
Processor: STM32F405

...

IMU: ICM42688 - cal_ax: cal_ay: cal_az: cal_gx: cal_gy: cal_gz: spi_bus:0 i2c_bus:-1 i2c_adr:0 align:CW270 rate: acc_lp: gyr_lp: bus_type:SPI 
IMU: detected:ICM42688P sample_rate:1000
IMU: Actual sample rate:1000 Hz
IMU: IMU_EXEC_IRQ
AHR: No Magnetometer, yaw:0.00
FATAL ERROR: IMU interrupt not firing. Is pin 'pin_imu_int' connected? Use CLI or reboot.
```

The IMU with interrupt works fine with [this ICM42688 library ](https://github.com/finani/ICM42688) using his Interrupt_SPI example.

I'm not savvy with c++ but from some poking around it seems the betaflight driver in madflight for the ICM42688P does not set up the interrupt the same way as that library. I am not familiar with writeRegister stuff and cannot tell which register is the issue. I might be wrong but I thought I read somewhere that Betaflight doesn't use the IMU interrupts anymore so this not working wouldn't matter to them but I'm not 100% sure if I am correct about that.

qqqlab, 2025-09-16:
Thanks for reporting this.

As far as I know most (if not all) Betaflight targets have a IMU interrupt pin. Also your board:

brd/betaflight/SPBE-SPEEDYBEEF405MINI.h

`pin_imu_int PC4 // resource GYRO_EXTI 1 C04`

However, I don't have this board, so I can't help you further... The sensor is detected, that is not the problem...

EnzoPB, 2025-09-16:
I can confirm this isn't working for the Mamba F405-mini and DFR F722 Dual HD, both using the ICM42688P. I've tried going back to older version of MF and it also doesn't work on version 2.0.1, version at which the ICM42688 gizmo was added.
I'm going to try to see what's wrong since I have the hardware to test it

EnzoPB, 2025-09-18:
Seems like the problem is related to STM32FreeRTOS. When I remove it from madflight (alongside with the bbx, rcl and lua modules, since they use FreeRTOS) the interrupt is getting called correctly.

Putting back only the `#include <STM32FreeRTOS.h>` in `hal/STM32/hal_STM32.h` breaks the interrupt.

The only similar case I've found is [this one](https://www.stm32duino.com/viewtopic.php?p=9737#p9737).

EnzoPB, 2025-09-18:
Setting the interrupt to a higher priority works:
(in imu/imu.cpp)
```cpp
void _imu_ll_interrupt_setup(int interrupt_pin) {
    Serial.println(MF_MOD ": IMU_EXEC_IRQ");
    attachInterrupt(digitalPinToInterrupt(interrupt_pin), _imu_ll_interrupt_handler, RISING);
    HAL_NVIC_SetPriority(EXTI4_IRQn, 0, 5); // set the subpriority to 5 (default is 6) for interrupt EXTI4 (pin PC4)
  }
```

If this fix were to be implemented, it still needs to:
- determine which HAL interrupt corresponds to interrupt_pin (`EXTI*_IRQn`)
- determine what priority level it needs to be (the subpriority 5 here is arbitrary)
- find out if it breaks something else (other interrupts, FreeRTOS tasks)